Kashif-E/Kamera #2: kotlin.native.internal.IrLinkageError: Function 'clickable' can not be called

kylixeza, 2024-09-22:
I always face this error when I use this library. All I did was import the library and run the app in iOS simulator without using any functions from the library. Import to gradle and sync only

```kotlin
kotlin.native.internal.IrLinkageError: Function 'clickable' can not be called: No function found for symbol 'androidx.compose.foundation/clickable|clickable@androidx.compose.ui.Modifier(androidx.compose.foundation.interaction.MutableInteractionSource;androidx.compose.foundation.Indication?;kotlin.Boolean;kotlin.String?;androidx.compose.ui.semantics.Role?;kotlin.Function0<kotlin.Unit>){}[0]'
Uncaught Kotlin exception:     at 0   Beu KMM                             0x101290acf        kfun:kotlin.Throwable#<init>(kotlin.String?){} + 119 
    at 1   Beu KMM                             0x101289c83        kfun:kotlin.Error#<init>(kotlin.String?){} + 115 
    at 2   Beu KMM                             0x1012c523b        kfun:kotlin.native.internal.IrLinkageError#<init>(kotlin.String?){} + 115 
    at 3   Beu KMM                             0x1012c52ff        kfun:kotlin.native.internal#ThrowIrLinkageError(kotlin.String?){}kotlin.Nothing + 175 
    at 4   Beu KMM                             0x1024c2717        kfun:androidx.compose.material#IconButton(kotlin.Function0<kotlin.Unit>;androidx.compose.ui.Modifier?;kotlin.Boolean;androidx.compose.foundation.interaction.MutableInteractionSource?;kotlin.Function2<androidx.compose.runtime.Composer,kotlin.Int,kotlin.Unit>;androidx.compose.runtime.Composer?;kotlin.Int;kotlin.Int){} + 3043 
    at 5   Beu KMM                             0x100c0798f        kfun:com.kylix.main.components#BottomNavItem(androidx.compose.ui.Modifier?;androidx.compose.ui.graphics.painter.Painter;kotlin.String;kotlin.Boolean;kotlin.Function0<kotlin.Unit>;androidx.compose.runtime.Composer?;kotlin.Int;kotlin.Int){} + 6007 
    at 6   Beu KMM                             0x100c060cf        kfun:com.kylix.main.components#TabNavigationItem__at__androidx.compose.foundation.layout.RowScope(cafe.adriel.voyager.navigator.tab.Tab;androidx.compose.runtime.Composer?;kotlin.Int){} + 2679 
    at 7   Beu KMM                             0x100c02f4f        kfun:com.kylix.main.MainScreen.Content$lambda$9$lambda$8$lambda$7$lambda$6#internal + 663 
    at 8   Beu KMM                             0x100c0467b        kfun:com.kylix.main.MainScreen.$Content$lambda$9$lambda$8$lambda$7$lambda$6$FUNCTION_REFERENCE$4.invoke#internal + 151 
    at 9   Beu KMM                             0x100c0481f        kfun:com.kylix.main.MainScreen.$Content$lambda$9$lambda$8$lambda$7$lambda$6$FUNCTION_REFERENCE$4.$<bridge-DNNNNU>invoke(androidx.compose.foundation.layout.RowScope;androidx.compose.runtime.Composer;kotlin.Int){}#internal + 195 
    at 10  Beu KMM                             0x1013db437        kfun:kotlin.Function3#invoke(1:0;1:1;1:2){}1:3-trampoline + 123 
    at 11  Beu KMM                             0x101685133        kfun:androidx.compose.runtime.internal.ComposableLambdaImpl#invoke(kotlin.Any?;androidx.compose.runtime.Composer;kotlin.Int){}kotlin.Any? + 635 
    at 12  Beu KMM                             0x10169393f        kfun:androidx.compose.runtime.internal.ComposableLambdaImpl#$<bridge-NNNNNU>invoke(kotlin.Any?;androidx.compose.runtime.Composer;kotlin.Int){}kotlin.Any?(kotlin.Any?;androidx.compose.runtime.Composer;kotlin.Any?){}kotlin.Any? + 199 
    at 13  Beu KMM                             0x1013db437        kfun:kotlin.Function3#invoke(1:0;1:1;1:2){}1:3-trampoline + 123 
    at 14  Beu KMM                             0x10249b507        kfun:androidx.compose.material.BottomNavigation$lambda$0#internal + 3771 
    at 15  Beu KMM                             0x10249bbcb        kfun:androidx.compose.material.$BottomNavigation$lambda$0$FUNCTION_REFERENCE$0.invoke#internal + 119 
    at 16  Beu KMM                             0x10249bd1f        kfun:androidx.compose.material.$BottomNavigation$lambda$0$FUNCTION_REFERENCE$0.$<bridge-DNNNU>invoke(androidx.compose.runtime.Composer;kotlin.Int){}#internal + 159 
    at 17  Beu KMM                             0x1013db33b        kfun:kotlin.Function2#invoke(1:0;1:1){}1:2-trampoline + 115 
    at 18  Beu KMM                             0x101684dc7        kfun:androidx.compose.runtime.internal.ComposableLambdaImpl#invoke(androidx.compose.runtime.Composer;kotlin.Int){}kotlin.Any? + 603 
    at 19  Beu KMM                             0x10169383b        kfun:androidx.compose.runtime.internal.ComposableLambdaImpl#$<bridge-NNNNU>invoke(androidx.compose.runtime.Composer;kotlin.Int){}kotlin.Any?(androidx.compose.runtime.Composer;kotlin.Any?){}kotlin.Any? + 163 
    at 20  Beu KMM                             0x1013db33b        kfun:kotlin.Function2#invoke(1:0;1:1){}1:2-trampoline + 115 
    at 21  Beu KMM                             0x102508ef7        kfun:androidx.compose.material.Surface$lambda$2#internal + 4327 
    at 22  Beu KMM                             0x102509db7        kfun:androidx.compose.material.$Surface$lambda$2$FUNCTION_REFERENCE$0.invoke#internal + 139 
    at 23  Beu KMM                             0x102509fd3        kfun:androidx.compose.material.$Surface$lambda$2$FUNCTION_REFERENCE$0.$<bridge-DNNNU>invoke(androidx.compose.runtime.Composer;kotlin.Int){}#internal + 159 
    at 24  Beu KMM                             0x1013db33b        kfun:kotlin.Function2#invoke(1:0;1:1){}1:2-trampoline + 115 
    at 25  Beu KMM                             0x101684dc7        kfun:androidx.compose.runtime.internal.ComposableLambdaImpl#invoke(androidx.compose.runtime.Composer;kotlin.Int){}kotlin.Any? + 603 
    at 26  Beu KMM                             0x10169383b        kfun:androidx.compose.runtime.internal.ComposableLambdaImpl#$<bridge-NNNNU>invoke(androidx.compose.runtime.Composer;kotlin.Int){}kotlin.Any?(androidx.compose.runtime.Composer;kotlin.Any?){}kotlin.Any? + 163 
    at 27  Beu KMM                             0x1013db33b        kfun:kotlin.Function2#invoke(1:0;1:1){}1:2-trampoline + 115 
    at 28  Beu KMM                             0x101578713        kfun:androidx.compose.runtime#CompositionLocalProvider(kotlin.Array<out|androidx.compose.runtime.ProvidedValue<*>>...;kotlin.Function2<androidx.compose.runtime.Composer,kotlin.Int,kotlin.Unit>;androidx.compose.runtime.Composer?;kotlin.Int){} + 615 
    at 29  Beu KMM                             0x102505aa7        kfun:androidx.compose.material#Surface(androidx.compose.ui.Modifier?;androidx.compose.ui.graphics.Shape?;androidx.compose.ui.graphics.Color;androidx.compose.ui.graphics.Color;androidx.compose.foundation.BorderStroke?;androidx.compose.ui.unit.Dp;kotlin.Function2<androidx.compose.runtime.Composer,kotlin.Int,kotlin.Unit>;androidx.compose.runtime.Composer?;kotlin.Int;kotlin.Int){} + 4883 
    at 30  Beu KMM                             0x10249931b        kfun:androidx.compose.material#BottomNavigation(androidx.compose.foundation.layout.WindowInsets;androidx.compose.ui.Modifier?;androidx.compose.ui.graphics.Color;androidx.compose.ui.graphics.Color;androidx.compose.ui.unit.Dp;kotlin.Function3<androidx.compose.foundation.layout.RowScope,androidx.compose.runtime.Composer,kotlin.Int,kotlin.Unit>;androidx.compose.runtime.Composer?;kotlin.Int;kotlin.Int){} + 3827 
    at 31  Beu KMM                             0x10249a243        kfun:androidx.compose.material#BottomNavigation(androidx.compose.ui.Modifier?;androidx.compose.ui.graphics.Color;androidx.compose.ui.graphics.Color;androidx.compose.ui.unit.Dp;kotlin.Function3<androidx.compose.foundation.layout.RowScope,androidx.compose.runtime.Composer,kotlin.Int,kotlin.Unit>;androidx.compose.runtime.Composer?;kotlin.Int;kotlin.Int){} + 3467 
    at 32  Beu KMM                             0x100c033bb        kfun:com.kylix.main.MainScreen.Content$lambda$9$lambda$8$lambda$7#internal + 843 
    at 33  Beu KMM                             0x100c04b63        kfun:com.kylix.main.MainScreen.$Content$lambda$9$lambda$8$lambda$7$FUNCTION_REFERENCE$7.invoke#internal + 151 
    at 34  Beu KMM                             0x100c04d07        kfun:com.kylix.main.MainScreen.$Content$lambda$9$lambda$8$lambda$7$FUNCTION_REFERENCE$7.$<bridge-DNNNNU>invoke(androidx.compose.animation.AnimatedVisibilityScope;androidx.compose.runtime.Composer;kotlin.Int){}#internal + 195 
    at 35  Beu KMM                             0x1013db437        kfun:kotlin.Function3#invoke(1:0;1:1;1:2){}1:3-trampoline + 123 
[... cut ...]
    at 124 Beu KMM                             0x10225b633        kfun:androidx.compose.animation.AnimatedEnterExitMeasurePolicy.measure#internal + 971 
    at 125 Beu KMM                             0x1021743ff        kfun:androidx.compose.ui.layout.MeasurePolicy#measure__at__androidx.compose.ui.layout.MeasureScope(kotlin.collections.List<androidx.compose.ui.layout.Measurable>;androidx.compose.ui.unit.Constraints){}androidx.compose.ui.layout.MeasureResult-trampoline + 123 

```
kotlin = "2.0.22"
compose plugin = "1.6.11"

Kashif-E, 2024-09-22:
have you tried with compose 1.7? clickable is not used inside the library but ill check for the root cause. do you have a reproducer?

kylixeza, 2024-09-22:
ahhh yes, upgrading compose plugin version to 1.7.0-beta02 solved this issue. Thanks man! @Kashif-E 👍🏻

you can close this issue 🙏🏻

Kashif-E, 2024-09-22:
that's great @kylixeza

closed 2024-09-22 by Kashif-E