Kashif-E/Kamera #8: Scanning the QR code will only emit to Flow when the QR code image is removed from the screen in IOS

adamglin0, 2024-10-07:


qq2540040142, 2024-10-14:
Android has the same problem

qq2540040142, 2024-10-14:
After checking, the problem is Flow.debounce(debounce)
'''
~~ ~~qrScannerPlugin.getQrCodeFlow(500)~~ ~~
qrScannerPlugin.getQrCodeFlow(0)
'''
In the example given, the debounce value is 500, which may cause the latest data to not be sent out in time because it is filtered out. Set it to 0, and now the QR code content can be obtained immediately.

adamglin0, 2024-10-14:
> After checking, the problem is Flow.debounce(debounce)
> 
> ```
> qrScannerPlugin.getQrCodeFlow(debounce = 0)
> ```
> 
> In the example given, the debounce value is **500**, which may cause the latest data to not be sent out in time because it is filtered out. Set it to **0**, and now the QR code content can be obtained immediately.

Yes, if the QR code remains on the screen, the debounce will continue to be triggered until 500ms after the screen no longer recognizes the QR code.

It would be ideal to add distinctUntilChanged before debounce.

Kashif-E, 2024-10-14:
@adamglin0 nice idea but this is a different issue on ios and flow issue is a different one. I guess rather then using flow it should be an atomic operation ill look into it more and then update the issue

PR #23 (2025-01-07): - fixed camera rotation
fix #18 
fix #16 
fix #8 

This pull request includes significant changes to the `ImageSaverPlugin` and `cameraK` modules, primarily focusing on improving image saving functionality and adding new features. The most important changes include updating the `ImageSaverPlugin` for iOS and Android, modifying the `CameraController` to support image rotation, and updating package names for consistency.

### ImageSaverPlugin improvements:

* [`ImageSaverPlugin/src/appleMain/kotlin/com/kashif/imagesaverplugin/ImageSaverPlugin.apple.kt`](diffhunk://#diff-22918b7548b2104becd33f58a [... cut]
closed 2025-01-07 by Kashif-E