Kashif-E/Kamera #15: Provide Sample Layout with icons only

MathiasDL, 2024-10-21:
For Android and iOS, the default camera view [uses icons](https://github.com/google/jetpack-camera-app/blob/main/README.md) for Capture, Flash, Back/front view.

The current sample uses Text Buttons instead. I would advise to also add a view with icons only, which you can even incorporate into the library itself as a default view. I would think most developers would go with such a default view, reducing the amount of code needed to implement this library. This has as an added bonus that you do not need to deal with localization, and you can also add a screenshot to the Readme.

Kashif-E, 2024-10-24:
@MathiasDL can you share a sample? so I can understand this better an image would work

MathiasDL, 2024-10-24:
See the link I posted, or a screenshot of the default camera view on my Samsung phone, relevant icons circled. As you can see, no text is required:

[image]

Kashif-E, 2024-10-24:
got ya, ill try to this asap

Kashif-E, 2024-10-24:
thanks @MathiasDL, I really appreciate that

PR #27 (2025-01-16): Enhancements and fixes
This pull request includes several changes across multiple files to refactor and simplify the codebase, particularly by removing the `Rotation` enum and related functionality. Additionally, there are some improvements to the `ImageSaverPlugin` and updates to the `README.MD` file.

fixes #26 
fixes #15 

### Refactoring and Simplification:

* [`cameraK/src/androidMain/kotlin/com/kashif/cameraK/builder/CameraControllerBuilder.android.kt`](diffhunk://#diff-1205132d59e8c63214ea264f2b62fdc30c24ac8e7fe4ad7f73b9fcd3dc2ca45eL29): Removed the `rotation` property and related metho [... cut]
closed 2025-01-16 by Kashif-E