Kashif-E/Kamera #16: Image URI

tanvirovi, 2024-10-24:
Get the image URI when saving the Image

Kashif-E, 2024-10-24:
@tanvirovi  thanks for the feature request, can you share a usecase for this?

tanvirovi, 2024-10-24:
@Kashif-E Efficient access to shared media files through the MediaStore. It also helps a lot with displaying all the saved images later. Storing URIs in a database can be helpful when you need to persist the reference to a media file across app sessions without storing the entire image file in the database.

Kashif-E, 2024-10-24:
Thanks @tanvirovi for explaining, this makes sense. I will try build this asap.

tanvirovi, 2024-10-24:
There are few useable camera libraries for KMP, but CameraK stands out. Thank you for the quick response.

Kashif-E, 2024-10-24:
Thanks @tanvirovi that's the goal

PR #23 (2025-01-07): - fixed camera rotation
fix #18 
fix #16 
fix #8 

This pull request includes significant changes to the `ImageSaverPlugin` and `cameraK` modules, primarily focusing on improving image saving functionality and adding new features. The most important changes include updating the `ImageSaverPlugin` for iOS and Android, modifying the `CameraController` to support image rotation, and updating package names for consistency.

### ImageSaverPlugin improvements:

* [`ImageSaverPlugin/src/appleMain/kotlin/com/kashif/imagesaverplugin/ImageSaverPlugin.apple.kt`](diffhunk://#diff-22918b7548b2104becd33f58a [... cut]
closed 2025-01-07 by Kashif-E