Kashif-E/Kamera #18: Images are rotated 90 degrees after capture on Android

RaedGhazal, 2024-11-23:
When I capture an image on Android the produced image is rotated 90 degrees no matter the Rotation configuration, on iPhone it works fine though.

Tested on: Pixel 8 pro, Pixel 6a, and samsung s23 ultra.

vide: https://drive.google.com/file/d/18800sMrycA-a-STcLOG6A20CgBbi4GBJ/view?usp=sharing

stasSuhanov, 2024-11-24:
Hi! Same issue.
Tested on: Pixel 9 pro, Pixel 7a.

Kashif-E, 2024-11-27:
I have reproduced the issue thanks for taking time out for testing and the video. Will try to fix this asap

RaedGhazal, 2024-12-16:
Please check this fix by a friend of mine on another library that had a similar issue, it could be helpful:
https://github.com/MoyuruAizawa/Cropify/pull/26

Kashif-E, 2024-12-16:
Hey @RaedGhazal  , apologies for the delay. It's a fairly common issue in Android, and I already know the fix. I've been traveling a lot lately, but I'll make sure to address it this week.

PR #23 (2025-01-07): - fixed camera rotation
fix #18 
fix #16 
fix #8 

This pull request includes significant changes to the `ImageSaverPlugin` and `cameraK` modules, primarily focusing on improving image saving functionality and adding new features. The most important changes include updating the `ImageSaverPlugin` for iOS and Android, modifying the `CameraController` to support image rotation, and updating package names for consistency.

### ImageSaverPlugin improvements:

* [`ImageSaverPlugin/src/appleMain/kotlin/com/kashif/imagesaverplugin/ImageSaverPlugin.apple.kt`](diffhunk://#diff-22918b7548b2104becd33f58a [... cut]
closed 2025-01-07 by Kashif-E