Kashif-E/Kamera #25: It looks like the camera rotation issue that was fixed in #23 is still present in 0.0.8 release

weedb, 2025-01-15:
I'm sure I'm using version `0.0.8`

If I need to provide any other information, please ask. 

here is the code i use:

```
val cameraController = remember { mutableStateOf<CameraController?>(null) }

CameraPreview(modifier = Modifier.fillMaxSize(), cameraConfiguration = {
    setCameraLens(CameraLens.BACK)
    setFlashMode(FlashMode.AUTO)
    setImageFormat(ImageFormat.JPEG)
    setDirectory(Directory.PICTURES)
}, onCameraControllerReady = { cameraController.value = it })

cameraController.value?.let { controller ->
    Box(Modifier.fillMaxSize()) {
        Image(imageVector = when(controller.getFlashMode()) {
            FlashMode.ON -> BurmistrIcons.FlashOn
            FlashMode.OFF -> BurmistrIcons.FlashOff
            else -> BurmistrIcons.FlashAuto
        }, contentDescription = null, colorFilter = ColorFilter.tint(Color.White), modifier = Modifier.align(Alignment.BottomStart).largeBottomPadding().mediumStartPadding().size(40.dp).noRippleClickable { errorLoadingScope { controller.toggleFlashMode() } })

        Image(imageVector = BurmistrIcons.Camera, contentDescription = null, colorFilter = ColorFilter.tint(Color.White), modifier = Modifier.align(Alignment.BottomCenter).largeBottomPadding().size(60.dp).noRippleClickable {
            errorLoadingScope {
                when (val result = controller.takePicture()) {
                    is ImageCaptureResult.Success -> { onCapture(result.byteArray) }
                    is ImageCaptureResult.Error -> {
                    }
                }
            }
        })

        Image(imageVector = BurmistrIcons.RefreshCcw, contentDescription = null, colorFilter = ColorFilter.tint(Color.White), modifier = Modifier.align(Alignment.BottomEnd).largeBottomPadding().mediumEndPadding().size(40.dp).noRippleClickable { errorLoadingScope { controller.toggleCameraLens() } })
    }
}
```

Kashif-E, 2025-01-15:
you need to set `setRotation(Rotation.ROTATION_180)` which ever rotation you want in camera configuration

weedb, 2025-01-15:
> you need to set `setRotation(Rotation.ROTATION_180)` which ever rotation you want in camera configuration

Thanks! This worked, I didn't try it initially because I thought it should work a little differently

closed 2025-01-15 by weedb

Kashif-E, 2025-01-15:
@weedb can you share what came to mind? it will help in making future design decisions

weedb, 2025-01-15:
@Kashif-E 

well, i expected that `Rotation` would be handled by the library if it is not specified manually.

if i understand correctly, now i will need to manually track the orientation of the device and change `Rotation` before taking a picture

as far as i understand, there is no such problem on ios at all (i can't check, because the library on `ios` doesn't work for me now), even without specifying `Rotation`, so i expected that if you don't specify `Rotation` in the common code, then on `android` this problem won't happen either