Kashif-E/Kamera #26: ios - Terminating app due to uncaught exception 'UIViewControllerHierarchyInconsistency'

weedb, 2025-01-15:
Hi! I have a problem with ios, when I try to use CameraPreview it gives me this exception

Terminating app due to uncaught exception `'UIViewControllerHierarchyInconsistency',` reason: 'child view controller:`<ComposeAppcom.kashif.cameraK.controller.CameraController12: 0x10b3f8f60>` should have parent view controller:(null) but actual parent is:`<ComposeHostingViewController: 0x10c009740>`'

camerak = "0.0.8"
compose = "1.7.1"
kotlin = "2.1.0-RC"

```
val cameraController = remember { mutableStateOf<CameraController?>(null) }

CameraPreview(modifier = Modifier.fillMaxSize(), cameraConfiguration = {
    setRotation(Rotation.ROTATION_90)
    setCameraLens(CameraLens.BACK)
    setFlashMode(FlashMode.AUTO)
    setImageFormat(ImageFormat.JPEG)
    setDirectory(Directory.PICTURES)
}, onCameraControllerReady = { cameraController.value = it })

cameraController.value?.let { controller ->
    Box(Modifier.fillMaxSize(), contentAlignment = Alignment.BottomCenter) {
        Row(Modifier.largeBottomPadding(), verticalAlignment = Alignment.CenterVertically, horizontalArrangement = Arrangement.spacedBy(50.dp)) {
            Image(imageVector = BurmistrIcons.Camera, contentDescription = null, colorFilter = ColorFilter.tint(Color.White), modifier = Modifier.size(60.dp).noRippleClickable {
                errorLoadingScope {
                    when (val result = controller.takePicture()) {
                        is ImageCaptureResult.Success -> { onCapture(result.byteArray) }
                        is ImageCaptureResult.Error -> { }
                    }
                }
            })
        }
    }
}
```

Kashif-E, 2025-01-16:
@weedb  cna you check if you still face the issue its related to compose how it works make sure you are not using viewcontroller when its already destroyed

closed 2025-01-16 by Kashif-E

weedb, 2025-01-16:
@Kashif-E I moved `CameraPreview` higher in the component tree and everything worked! Thanks!

weedb, 2025-01-16:
@Kashif-E well, i didn't use swift ui, only compose, i don't know why i got such an error, maybe because i opened the `CameraPreview` inside the popup

serhii-petrenko-dev, 2025-02-07:
Same thing happened to me. Steps to reproduce: CameraPreview inside ModalBottomSheet. Works on Android, crashes on IOS with error described above. Had to switch BottomSheetScaffold and make it behave like ModalBottomSheet.

PR #27 (2025-01-16): Enhancements and fixes
This pull request includes several changes across multiple files to refactor and simplify the codebase, particularly by removing the `Rotation` enum and related functionality. Additionally, there are some improvements to the `ImageSaverPlugin` and updates to the `README.MD` file.

fixes #26 
fixes #15 

### Refactoring and Simplification:

* [`cameraK/src/androidMain/kotlin/com/kashif/cameraK/builder/CameraControllerBuilder.android.kt`](diffhunk://#diff-1205132d59e8c63214ea264f2b62fdc30c24ac8e7fe4ad7f73b9fcd3dc2ca45eL29): Removed the `rotation` property and related metho [... cut]