Kashif-E/Kamera #29: Make customCameraController internal

maciek-s, 2025-02-06:
Hello, it would be great to have access to the `customCameraController` field inside `CameraController`. I need access to the `AVCaptureVideoPreviewLayer` for my implementation. 
Thanks for your great work!”

Kashif-E, 2025-02-06:
@maciek-s could you share your usecase? maybe we could find a better way to handle it

maciek-s, 2025-02-06:
I think I was a bit rushed (I’m an Android developer with limited experience in iOS), but I found that I can extract it from `AVCaptureConnection`…

maciek-s, 2025-02-06:
Unfortunately `previewLayer` is always null... I would like to map scanned barcode `corners/bound` to camera preview position 
```
override fun captureOutput(
    output: AVCaptureOutput,
    didOutputMetadataObjects: List<*>,
    fromConnection: AVCaptureConnection
) {
    for (metadata in didOutputMetadataObjects) {
        if (metadata is AVMetadataMachineReadableCodeObject) {
            metadata.stringValue?.let { code ->
                fromConnection.videoPreviewLayer?.let { previewLayer ->
                }
                onResult(
                    CardBarcode(
                        code = code,
                        type = metadata.getCardCodeType()
                    )
                )
            }
        }
    }
}
```

Kashif-E, 2025-02-06:
you should use the plugin api to build your own plugin

Kashif-E, 2025-02-06:
are you working on a fork? could you share project link?

maciek-s, 2025-02-06:
This is not a fork, and I can’t share the project link as it is private. However, I manually calculated the necessary information based on the corner points from `metadata.corners` and my scanner area fraction points. I just need to map it based on the device orientation, which on my device is always `AVCaptureVideoOrientationLandscapeRight`. When I rotate, the preview becomes incorrect, as I reported [here](https://github.com/Kashif-E/CameraK/issues/30).

Kashif-E, 2025-02-06:
ill look into it will fix this in the next release

Kashif-E, 2025-02-14:
check the latest master

maciek-s, 2025-02-18:
Hi, thanks for the function to access the preview layer—it works!

closed 2025-02-18 by maciek-s