Kashif-E/Kamera #30: iOS wrong preview orientation in landscape

maciek-s, 2025-02-06:
On an iOS device, the camera preview has an incorrect orientation after the device is rotated to landscape mode.

Sischinger, 2025-02-08:
I encountered that issue as well for an iPhone 13mini, however the issue did not occur for an iPhone SE model

maciek-s, 2025-02-08:
This is my Xcode setting. I changed the orientation to landscape by rotating the device. On my iPhone 13 Mini, the camera preview is in full screen but has the wrong orientation.
[image]

Sischinger, 2025-02-08:
I think you have to listen to orientation changes and apply the correct videoOrientation in the customCameraController

maciek-s, 2025-02-08:
`customCameraController` is private, and I use this library directly, so it would be great if you could manage to fix this.

Kashif-E, 2025-02-14:
hey folks, @maciek-s @Sischinger  can   you the latest in master if you can reproduce this?

maciek-s, 2025-02-18:
Hi, I can confirm that rotation works on master. Thanks for the quick fix!

closed 2025-02-18 by Kashif-E