Kashif-E/Kamera #31: Add photo and preview format configuration to CameraControllerBuilder

serhii-petrenko-dev, 2025-02-08:
Hey, thanks for the great library!
I've encountered a limitation in my app where, after taking a photo on iOS, the captured image does not match the camera preview.

**Issue**
By default, cameraPreviewLayer?.videoGravity is set to .resizeAspectFill, and the default photo format is 16:9, while my preview was in 4:3. This mismatch caused inconsistencies between what was shown in the preview and the final captured image.

**Solution Attempt**
After multiple trial-and-error attempts, I found a way to set a custom sessionPreset to resolve this issue:
```
override fun initialize(controller: CameraController) {
      cameraController = controller
      val videoPreviewLayer = controller.view.layer.sublayers?.firstOrNull {
          it is AVCaptureVideoPreviewLayer
      } as? AVCaptureVideoPreviewLayer ?: return
      val captureSession = videoPreviewLayer.session ?: return
      captureSession.sessionPreset = AVCaptureSessionPresetPhoto // <--- custom preset
  }
```

It would be great to have an API in CameraControllerBuilder to configure both the image format and preview format before starting the capture session.

deividasstr, 2025-04-09:
Hey, I have the same requirement, where the captured image ratio/boundaries are different from the ones in preview, but also on android.

Would be nice to configure it. Or at least have a way to manage quality of the image (now I could crop image according to the screen size, but my screen turns out bigger (more px width and height) than the outcome `byteArray.decodeToImageBitmap()` (unless this is somehow expected?)

Kashif-E, 2025-04-09:
Will be taking this up in the next release to make this customisable and to manage quality as well

deividasstr, 2025-04-14:
If you need some help on this, just let me know what exact part I could tackle in code

Kashif-E, 2025-04-14:
I have started working on this and #44 will be releasing this over the weekend