Kashif-E/Kamera #32: Crash while capture photo in IOS

ravi2210, 2025-02-14:
1. App Crash
When we are doing fast photo capture then app is getting crash.

Steps to reproduce

continue click on capture button in ios app.It will crash immediately.

2. Image roation.

Once image got save in memory using image saver and try to load on UI from image path then image got rotate to diff angle in IOS. Android is working expected.
After saving image using Image saver

Kashif-E, 2025-02-14:
Hey @ravi2210 by different angle do you mean inverted image?

ravi2210, 2025-02-14:
Hey @Kashif-E  sorry the image rotation issue is fixed when I used coil lib to load the image. thanks for you support.

But for fast photo capture in IOS we need support. Can you please check on this.

Kashif-E, 2025-02-14:
sure @ravi2210  i am currently working on fixing some other issues as well, this fix will be available by the weekend

Kashif-E, 2025-02-14:
@ravi2210 would you be able to test the code in master if you are able to reproduce the issue?

ravi2210, 2025-02-14:
@Kashif-E  sure let me check, will update you asap

ravi2210, 2025-02-14:
I am getting this log "Image Capture Error: Capture too frequent", and after that I am not able to capture any other image.
How to enable multiple photo capture just like native camera..

Kashif-E, 2025-02-14:
which device are you on? @ravi2210   and your ios verion?

ravi2210, 2025-02-14:
IOS 15

Kashif-E, 2025-03-13:
Please ignore this. Just checking issue information.

PR #42 (2025-03-26): Fix ios rapid capture
closes #32 
This pull request includes significant updates to the `cameraK` module, including the addition of a new `BurstCaptureManager` class, updates to the `CameraController` class, and changes to the build configuration files to support these updates. Additionally, there are removals of certain plugins and functionalities in the `Sample` app.

### Major changes:

#### New Features:
* **Burst Capture Management:**
  * Added `BurstCaptureManager` class to handle burst mode captures efficiently, optimizing for rapid sequential photo captures with minimal delay. This inc [... cut]
closed 2025-03-26 by Kashif-E