Kashif-E/Kamera #41: Support voor zooming

NielsMasdorp, 2025-03-07:
Hi,

Do you plan to support zoom interaction?

Thanks

Kashif-E, 2025-03-07:
is voor supposed to be for?

jeacott1, 2025-05-02:
+1 for zoom

a-xnow, 2025-06-27:
+1 for zoom

licryle, 2026-02-21:
Thanks a lot!!

By the way, it seems like the documentation is outdated, I had to use the Sample rather than the doc to update the 0.2.0.

Also, might be nice to add a simple pinch to zoom example in the sample app !

```
        val zoomLevel = remember { mutableStateOf(1f) }

        CameraKScreen(
...
        ) { readyState ->
            Box(
                modifier = Modifier
                    .fillMaxSize()
                    .pointerInput(Unit) {
                        detectTransformGestures { _, _, zoomChange, _ ->
                            zoomLevel.value *= zoomChange
                            zoomLevel.value = zoomLevel.value.coerceIn(1f, 10f)
                            readyState.controller.setZoom(zoomLevel.value)
                        }
                    }
            ) {
                CaptureButton(
                    onClick = { viewModel.takePhoto(readyState.controller) },
                    modifier = Modifier
                        .align(Alignment.BottomCenter)
                        .padding(bottom = 50.dp)
                )
            }
        }
```

Thanks again!