Kashif-E/Kamera #43: How not to show the interface, just run in the background

jwli6, 2025-04-09:
I don't need to display the image interface, I just continue to collect images in the background(by thread) and call my own recognition algorithm for the images, what should I do?

Kashif-E, 2025-04-09:
@jwli6 when you click an image it doesnt get saved or gets shown directly the library provides a bytearray which you can use to  get a bitmap and run your algo on that

jwli6, 2025-04-09:
> [@jwli6](https://github.com/jwli6) when you click an image it doesnt get saved or gets shown directly the library provides a bytearray which you can use to get a bitmap and run your algo on that

first use CameraK by Composable function CameraPreview; (but, I don't want to show anything on the screen, just running in the background.)

then I'd like to register a callback function into the library.

The pseudocode is similar to the following:
```kotlin
fun myAlgo(img:ByteArray){
    // do something;
}

var camera = CameraK()
camera.addImageCaptureListener(myAlgo)

```

jwli6, 2025-04-14:
I wrote a kotlin multiplatform plugin to do the above function, but I encountered the following problem:
1. I don't know how to export the plugin locally, and test the plugin in another project, 
2. and finally if the test is successful, how to submit it to the current git project and push it to mavenCentral

Can someone help me? Please

Kashif-E, 2025-04-14:
@jwli6  you can run `./gradlew publishToMavenLocal` but if you have the library in. your project you dont need to publish and add it to your project you can move the plugin to your own project

jwli6, 2025-04-14:
> [@jwli6](https://github.com/jwli6) you can run `./gradlew publishToMavenLocal` but if you have the library in. your project you dont need to publish and add it to your project you can move the plugin to your own project

Thanks.

closed 2025-04-14 by jwli6