Kashif-E/Kamera #44: Image is saved in wrong orientation on IOS

jeremarques, 2025-04-09:
#### Hello! First of all, thanks for the great library!

I am developing an application that takes photos and saves them to Amazon S3 storage. However, in the **iOS application**, the image is saved in the wrong orientation, while in the Android application, the image is saved in the correct orientation.
For example, my photo was taken in **portrait** orientation, but it was saved in **landscape** orientation.

The version I am using is **0.0.9**. I tried updating to version **0.0.11**, but the bug persists and new bugs have been introduced.

jeremarques, 2025-04-09:
Thanks @Kashif-E  for the quick response.

1. I tested on two iPhones, one with iOS **15.7.8** and the other with iOS **18.3.2**.
2. My CMP version is **1.7.0**
3. The process I follow to send the image is:
    - Take the photo
    - Send the bytearray to my API to handle the compression and validation of the image
    - After that, I save the image to S3

In short, in this process what I do is basically **compress** and **save** the image in S3, but when I do the process in **IOS** and go to check in S3 the image is in the **wrong orientation**.

Kashif-E, 2025-04-09:
no compression on mobile side only sending the raw bytearray you get from the library?

jeremarques, 2025-04-09:
I had already tried to remove the compression, but the problem still persisted.

Hasan-rapido, 2025-04-14:
@Kashif-E , Is it possible to set the image resolution which we want to capture? and the captured image in android is mirrored in the preview. How do I solve this?

Kashif-E, 2025-04-14:
@Hasan-rapido currently its not possible this is targeted for next release, expect it to be available by the coming weekend

Hasan-rapido, 2025-04-14:
Thanks for the update @Kashif-E .
The selfie image captured is getting mirrored. How do I solve this?

Dwite, 2025-08-22:
@Kashif-E hey , just wanted to check if this was released? based on source it feels like the iOS is still stripping the EXIF and not rotating the image before providing it in callback in a way Android controller is doing?