Kashif-E/Kamera #48: [iOS][Bug] - First time capturing image from is coming is black image but subsequent calls are working fine.

SURYAKANTSHARMA, 2025-04-16:
- iOS version 18.3.2 (iphone 13)

- When using cameraController.takePicture(), the **first capture** from **front camera** returns a black image (likely empty or not yet ready), but subsequent captures work as expected.
Steps to reproduce on master branch:
	1.	launch app on iOS device 
	2.	Flip camera to front camera 
	3.     capture photo 
	4.	Observe that the first image is black
	5.	Retry and see subsequent images captured correctly

https://github.com/user-attachments/assets/162207cd-aae7-48ef-a561-ed9076f90f82
Sample App: https://github.com/Kashif-E/CameraK/tree/main/iosApp

SuryakantRapido, 2025-04-17:
@Kashif-E can you please look into this?

Kashif-E, 2025-04-17:
@SuryakantRapido which ios and cmp version are you on?

SURYAKANTSHARMA, 2025-04-17:
@Kashif-E iOS version 18.3.2.

Kashif-E, 2025-04-17:
Thanks will look into this look for an update by the weekend

SURYAKANTSHARMA, 2025-04-19:
@Kashif-E Looking forward to hear back. Let me know if you have any issue with reproduction steps.

closed 2025-11-27 by Kashif-E