Kashif-E/Kamera #59: long org.jetbrains.skiko.node.RenderNodeContextKt.RenderNodeContext_nMake(boolean)

amolaksingh, 2025-05-19:
I written below libraries in comman buildgradle and in commonMain.dependency{
          implementation("io.github.kashif-mehmood-km:camerak:+")
            implementation("io.github.kashif-mehmood-km:image_saver_plugin:0.0.7")
}

I am getting below error when I try to run in desktop

Exception in thread "main" java.lang.UnsatisfiedLinkError: 'long org.jetbrains.skiko.node.RenderNodeContextKt.RenderNodeContext_nMake(boolean)'

Kashif-E, 2025-05-19:
Hi @amolaksingh  are you using any plugin?

amolaksingh, 2025-05-20:
I am using below plugins
 

alias(libs.plugins.kotlinMultiplatform)
alias(libs.plugins.androidLibrary)
alias(libs.plugins.androidApplication)
alias(libs.plugins.composeMultiplatform)
alias(libs.plugins.composeCompiler)
id("org.jetbrains.compose.hot-reload") version "1.0.0-alpha02"

libs.toml
androidLibrary = { id = "com.android.library", version.ref = "8.10.0" }
androidApplication = { id = "com.android.application", version.ref = "8.10.0" }
kotlinMultiplatform = { id = "org.jetbrains.kotlin.multiplatform", version.ref = "2.1.20" }
composeMultiplatform = { id = "org.jetbrains.compose", version.ref = "1.7.3" }
composeCompiler = { id = "org.jetbrains.kotlin.plugin.compose", version.ref = "2.1.20" }

amolaksingh, 2025-05-22:
@Kashif-E Any Solution for this issue?

MichaelJob, 2025-05-23:
I experience the same error: 
with java/21.0.4-tem (M1)
Exception in thread "main" java.lang.UnsatisfiedLinkError: 'long org.jetbrains.skiko.node.RenderNodeContextKt.RenderNodeContext_nMake(boolean)'

Kashif-E, 2025-05-23:
hey folks ill look into this tomorrow